swri-robotics/novatel_gps_driver #6: Insufficient data rate is logged for GPS rates lower than 20 Hz

pjreed, 2017-09-11:
A user reported seeing this message logged when using an OEM718D device that was running at 5 Hz.  Looking at the code, I suspect that this warning will be logged any time GPS messages are received at a rate of less than 20 Hz, even if the `polling_period` is configured differently.

olamarre, 2018-07-06:
Hi! I am using a Novatel Smart6L, and I am also getting this error. Has anyone figured out a fix for it?
Also, I am not receiving any data on the published topics (except for diagnostics). Is this related to this error?
Thank you!

senoa95, 2018-07-27:
Hi @olamarre sorry for the delay, I was not able to get to this earlier. It seems the wizard is configured correctly. I assume that you are using the 'tester_for_usb.launch' to launch the driver. Have you tried changing the device to use one of the other USB ports eg. try (device: /dev/ttyUSB1 or 
device: /dev/ttyUSB2). I hope that fixes your issue.

Additionally, I think one of the latest commits might have fixed the issue but there are some carriage returns ('\r') missing in the novatel_gps.cpp file in the novatel_gps_driver directory. Give that a shot. 

Good luck!

olamarre, 2018-08-23:
Hi @senoa95 ! No worries. I was actually off the grid for the past month - I'll try it and let you know if I have further problems. Thanks a lot!

weisongwen, 2019-03-09:
@olamarre  have you solve the probelm? i got the same probelm,

mrgransky, 2019-12-06:
Still the same error:

`[ERROR] [1575627506.957520432]: insufficient data rate <Novatel GPS (/dev/ttyUSB0)>: 0.000000 < 20.000000`

and tried with all three `/dev/ttyUSB0` , `/dev/ttyUSB1`, `/dev/ttyUSB2` and my device: `pwrpack7`.

$ rostopic list -v 

```
Published topics:
 * /clocksteering [novatel_gps_msgs/ClockSteering] 1 publisher
 * /inspvax [novatel_gps_msgs/Inspvax] 1 publisher
 * /fix [sensor_msgs/NavSatFix] 1 publisher
 * /heading2 [novatel_gps_msgs/NovatelHeading2] 1 publisher
 * /gpgga [novatel_gps_msgs/Gpgga] 1 publisher
 * /corrimudata [novatel_gps_msgs/NovatelCorrectedImuData] 1 publisher
 * /gprmc [novatel_gps_msgs/Gprmc] 1 publisher
 * /time [novatel_gps_msgs/Time] 1 publisher
 * /rosout [rosgraph_msgs/Log] 1 publisher
 * /inscov [novatel_gps_msgs/Inscov] 1 publisher
 * /rosout_agg [rosgraph_msgs/Log] 1 publisher
 * /bestpos [novatel_gps_msgs/NovatelPosition] 1 publisher
 * /imu [sensor_msgs/Imu] 1 publisher
 * /insstdev [novatel_gps_msgs/Insstdev] 1 publisher
 * /bestxyz [novatel_gps_msgs/NovatelXYZ] 1 publisher
 * /inspva [novatel_gps_msgs/Inspva] 1 publisher
 * /range [novatel_gps_msgs/Range] 1 publisher
 * /diagnostics [diagnostic_msgs/DiagnosticArray] 1 publisher
 * /bestutm [novatel_gps_msgs/NovatelUtmPosition] 1 publisher
 * /gps [gps_common/GPSFix] 1 publisher
 * /bestvel [novatel_gps_msgs/NovatelVelocity] 1 publisher

Subscribed topics:
 * /gps_sync [std_msgs/Time] 1 subscriber
 * /rosout [rosgraph_msgs/Log] 1 subscriber
```
but neither one of them has sensor reading!

pjreed, 2019-12-06:
@mrgransky , I think your issue is unrelated to this.  This issue is because the diagnostics will log a warning if your data rate is less than 20 Hz even if you have actually configured your receiver to produce less than that, but your issue seems to be that you're not actually publishing data at all.  If you've built the driver from the latest source and have addressed the comments I made in #75 and #18, and you're still seeing this problem, could you open a new issue and paste a log file from when you run the driver?

senoa95, 2019-12-06:
@mrgransky I have run into a similar issue using Novatel Span CPT. It is possible that the ROS driver is not communicating with the novatel hardware. Please look at my [ comment above ](https://github.com/swri-robotics/novatel_gps_driver/issues/6#issuecomment-403187704) to see how I solved this issue, basically by making all received messages of type "Novatel".

mrgransky, 2019-12-09:
Thanks for your support. The problem was indeed connection with GPS antenna. Now it publishes data with sensor readings! 
And I built it from the source.

closed 2026-09-13 by danthony06