swri-robotics/novatel_gps_driver #12: Ability to send configuration commands to Novatel

mayurshah530, 2017-11-17:
Existing release translates the rosparams to various logs in NovatelGps::Configure. There is no way to send configuration messages to Novatel. (e.g. connectimu, setimuorientation etc.)
It would be great if this driver could support functionality to set configuration that is not log + x + ontime + y

evenator, 2017-12-06:
Are you imagining a parameter to send arbitrary commands at config time or a service to send arbitrary commands while running?

mayurshah530, 2017-12-06:
Hi Edward,
I was thinking of the initial configuration. While Novatel receivers allow
for saveconfig command that will be persisted, it is always more robust to
configure things per the mission request - especially if the hardware is
shared and used for different purposes / different configurations. At the
moment, I have to hook it up to a windows laptop with Novatel Connect or
open the web UI to send different configuration commands manually.
If you look at the http://wiki.ros.org/novatel_span_driver, it was able to
send configuration commands.
Best,
Mayur

closed 2026-09-20 by danthony06