swri-robotics/novatel_gps_driver #13: Cmake linking fail on boost

StevenCoral, 2018-01-01:
Received error message on end of build:
[100%] Built target driver_nodelet
/home/username/catkin_ws/devel/lib/libnovatel_gps_driver_nodelets.so: undefined reference to `boost::this_thread::hidden::sleep_until(timespec const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/username/catkin_ws/devel/lib/novatel_gps_driver/novatel_gps_node] Error 1
make[1]: *** [novatel_gps_driver/CMakeFiles/novatel_gps_node.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

pjreed, 2018-01-02:
Could you also share the following:
1) Operating system version
2) Commands used to create & build the workspace
3) Version of boost installed

Thanks!

StevenCoral, 2018-01-02:
Ubuntu 14.04
boost 1.65 and 1.54
Command issued is a catkin_make without any changes.
In addition, following the exact instructions on this git step by step, catkin build gave me the same error but preceded with something like "boost version 1.54 might cause contradiction with the version 1.65 you have installed", so I downgraded to 1.54 and got the same error without this sentence.
Installing from apt-get though worked fine.

pjreed, 2018-01-03:
If you had installed boost 1.65 from source, I suspect there are still headers or cmake files somewhere that are causing conflicts.  In particular, I see that the errors you pasted are referencing this symbol:

```
/home/username/catkin_ws/devel/lib/libnovatel_gps_driver_nodelets.so: undefined reference to `boost::this_thread::hidden::sleep_until(timespec const&)'
```

But that doesn't actually exist in Boost 1.54.0 on Ubuntu 14.04:

```
$ nm -C -D /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0 |grep sleep_until
000000000000c540 T boost::this_thread::hiden::sleep_until(timespec const&)
```

Note the typo in there (**hiden**) which was probably fixed at some point between 1.54 and 1.65.

pjreed, 2019-02-26:
I'm going to close this issue since it hasn't been updated in a long time.

closed 2019-02-26 by pjreed