swri-robotics/novatel_gps_driver #16: How to get bestpos and bestvel from ttyUSB0

jones5322, 2018-03-11:
After building from instructions using catkin I used `$ roslaunch /home/andrew/novatel/src/novatel_gps_driver/novatel_gps_driver/launch/tester_for_usb.launch` to launch the unedited launch file.  The error I received back was `Failed to load nodelet [/novatel] of type [novatel_gps_driver/novatel_gps_nodelet] ` with further details attached.  
[error.txt](https://github.com/swri-robotics/novatel_gps_driver/files/1800864/error.txt)  Can I get some guidance on how to change the tester_for_usb.launch file?
[tester_for_usb.txt](https://github.com/swri-robotics/novatel_gps_driver/files/1800868/tester_for_usb.txt)

No doubt you realize I'm a noob and the training I'm taking at robotignite academy so far has had fairly simple launch files and the packages and folder structure are dissimilar to yours.  My goal at the moment is simply to get the binary format coming into USB0 into ROS so I can echo out the results of BESTPOSB and BESTVELB.  The reason I used the roslaunch command the way I did is because I could not get the 2-3 different source commands I tried to let me follow the typical structure of roslaunch (package) (launch file).  Thanks for your help and appreciate that you guys have provided the driver.

pjreed, 2018-03-12:
Based on this line in your error log:
```
[ERROR] [1520799297.178989183]: The error before refreshing the cache was: According to the loaded plugin descriptions the class novatel_gps_driver/novatel_gps_nodelet with base class type nodelet::Nodelet does not exist. Declared types are  depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2 swri_image_util/blend_images swri_image_util/contrast_stretch swri_image_util/crosshairs swri_image_util/draw_polygon swri_image_util/draw_text swri_image_util/image_pub swri_image_util/normalize_response swri_image_util/replace_colors swri_image_util/rotate_image swri_image_util/scale_image swri_image_util/warp_image swri_nodelet/TestNodelet swri_transform_util/dynamic_publisher swri_transform_util/gps_transform_publisher
```
It looks like ROS wasn't able to find the built nodelet library, which makes me suspect that you didn't source your setup file (`. $HOME/novatel/devel/setup.bash`) after building the workspace.  Installing the binary package (`ros-kinetic-novatel-gps-driver`) would "fix" that problem because now you have the driver nodelet installed in /opt/ros and would not need to source the workspace in order for ROS to find it.

pjreed, 2019-11-13:
I think this has been fixed now, so I'll close this issue.

closed 2019-11-13 by pjreed