swri-robotics/novatel_gps_driver #18: Has anyone tested this with Novatel Pwrpak7?

vinhk, 2018-04-10:
I am using Pwrpak7 but I can not get IMU topic. I am hoping that I can get some help with making the IMU topic to display on the terminal.

Thank you

pjreed, 2018-04-10:
@JWhitleyAStuff actually just submitted a PR (#17) that adds support for some of the PowerPak7's IMUs.  If you're not building from the latest source, you might give that a try and see if it works better for you.

JWhitleyAStuff, 2018-04-10:
The IMU from the PowerPak 7 should be supported after that PR. However, we were still having issues with the parsing of INSSTDEV that we didn't have time to fix.

vinhk, 2018-04-10:
Yes, I added the parameter to the launch file as so: 

<?xml version="1.0"?>
<launch>
  <node name="novatel"
        pkg="nodelet" type="nodelet"
        args="standalone novatel_gps_driver/novatel_gps_nodelet">
    <rosparam>
      verbose: false
      connection_type: serial
      device: /dev/ttyUSB0
      publish_novatel_positions: true
      frame_id: /gps
	  imu_frame_id: /imu
	  imu_rate: 100
	  publish_imu: true
    </rosparam>
  </node>
</launch>

JWhitleyAStuff, 2018-04-10:
@vinhk - FYI, the publish rate for the PowerPak7's internal IMU (Epson G320N) is 125Hz.

JWhitleyAStuff, 2018-04-10:
@vinhk - We had hoped that the issue we were seeing was isolated to our firmware/specific unit but it seems that is not the case. We were seeing that, the second time we would connect to the unit, it would lose it's configuration (INS config, antenna offsets, etc.) and would have to be reconfigured. This causes only certain topics to be published with the driver and none of the IMU messages appear to be reported. You'll have to contact Novatel support to see if they can help you out. This driver isn't doing anything unusual. It calls UNLOGALL upon connection to make sure nothing unusual is being published and then LOG <message> <interval> several times to get the messages that the driver cares about. Doesn't even attempt to save the configuration.

vinhk, 2018-04-10:
@JWhitleyAStuff I see. Did it work the first time you connect it? Did the IMU topic get display? If that is the problem, then short term would be to leave the IMU system connected to the PC.

JWhitleyAStuff, 2018-04-10:
@vinhk - It published IMU data on first connect after configuration but it had problems parsing the INSSTDEV message which we did not have time to troubleshoot. Everything else seemed fine (on first connect).

vinhk, 2018-04-10:
@JWhitleyAStuff Thanks. We will work with that info for now. Until Novatel support provide the solution.

kelvin-sudani, 2018-09-10:
@vinhk  I am not sure about the parameter ` publish_imu: true` in your launch file, unless you have modified the source code with this parameter name. I think it should be `publish_imu_messages: true` in the launch file.

sharvil-patankar, 2019-09-24:
I don't know if this is still an issue but using binary instead of ASCII seemed to solve the problem for me

mrgransky, 2019-12-06:
> PowerPak7's internal IMU (Epson G320N) is 125Hz.

in `pwrpack7`, assigning `imu_rate: 125`, turns into this warning:

`[ WARN] [1575640680.489097734]: insufficient data rate <Novatel GPS (/dev/ttyUSB0)>: 17.000000 < 20.000000`

pjreed, 2019-12-06:
The "insufficient data rate" message can be logged by any publisher that is not successfully publishing at the requested data rate.  If it happens when you enable your IMU, check:
1. Are you using binary messages rather than ASCII?
2. If you're using a serial connection, what is the baud rate for your device?

IMU messages take a *lot* of bandwidth (relatively) and may be saturating your connection if you're using ASCII messages or a low baud rate connection.

JamesSWilliams, 2020-02-25:
Hey, I'm currently getting `Invalid ASCII char` with a PwrPak7D, not sure what I'm doing wrong. I've also set the port to output binary data `INTERFACEMODE CCOM1 NOVATELBINARY NOVATELBINARY ON`.
Any help would be appriciated.

Log:
```
[ WARN] [1582636236.426207649]: publish rate failures detected <Novatel GPS (192.168.43.180:3001)>: 1
[ WARN] [1582636236.426492873]: Novatel status code: 33554432
[ WARN] [1582636236.549404486]: INSPVA queue overflow.
[ WARN] [1582636236.641416087]: CORRIMUDATA queue overflow.
[ WARN] [1582636236.968506748]: Invalid ASCII char: [$???]
[ WARN] [1582636236.968623445]:                      ^
[ WARN] [1582636237.091957906]: Invalid ASCII char: [#????]
[ WARN] [1582636237.092083692]:                      ^
[ WARN] [1582636237.163826842]: Invalid ASCII char: [$?&?b?]
[ WARN] [1582636237.163949321]:                        ^
[ERROR] [1582636237.426488608]: insufficient data rate <Novatel GPS (192.168.43.180:3001)>: 0.000000 < 20.000000
[ WARN] [1582636237.426639386]: publish rate failures detected <Novatel GPS (192.168.43.180:3001)>: 1
[ WARN] [1582636237.426789687]: Novatel status code: 33554432
```

Launch file:
```yaml
<launch>
  <!--http://wiki.ros.org/novatel_gps_driver-->
  <node name="novatel"
        pkg="nodelet" type="nodelet"
        args="standalone novatel_gps_driver/novatel_gps_nodelet">
    <rosparam>
      connection_type: tcp
      device: "192.168.43.180:3001"
      verbose: true
      imu_sample_rate: -1
      use_binary_messages: true
      publish_novatel_positions: true
    </rosparam>
  </node>
</launch>
```

pjreed, 2020-02-25:
That's a little odd; it looks like you're not actually getting ASCII messages, but it's failing to identify where the binary messages are in the data stream an erroneously trying to parse any string that starts with a `#` or `$` as an ASCII message.

Something else suspicious is that in your config, you don't have `publish_imu_messages` set to `true`, but your log has warnings about the INSPVA / CORRIMUDATA buffers overflowing, and you shouldn't be getting logs of those type at all, so something is going wrong with the configuration.

I think that having manually run `INTERFACEMODE CCOM1 NOVATELBINARY NOVATELBINARY ON` may be causing problems; that also causes the device to only *accept* binary commands, but this driver submits all of its commands as ASCII.  Although, you might also not be configuring the port you think you are; as far as I'm aware, the `CCOM1` port is just for CAN bus communication, and TCP communication is done over the `ICOM` ports.  Can you double-check which device is listening on TCP port 3001?  (`LOG ICOMCONFIG ONCE`)  If that device is in `NOVATELBINARY` mode, it should probably be in `NOVATEL` mode instead.

JamesSWilliams, 2020-02-25:
Sorry for the confusion,  I've set on `ICOM1` to `NOVATELBINARY`, must have copied it wrong...
I've changed `ICOM1` back to `NOVATEL`, now it seems to work, `/bestpos`  is published with the correct position

I still get these warnings & errors.
```
[ WARN] [1582650331.539180895]: publish rate failures detected <Novatel GPS (192.168.43.180:3001)>: 2
[ WARN] [1582650331.539365724]: Novatel status code: 33554432
[ERROR] [1582650332.571151664]: insufficient data rate <Novatel GPS (192.168.43.180:3001)>: 4.000000 < 20.000000
[ WARN] [1582650332.571259911]: publish rate failures detected <Novatel GPS (192.168.43.180:3001)>: 1
[ WARN] [1582650332.571380400]: Novatel status code: 33554432
[ WARN] [1582650333.613886302]: publish rate failures detected <Novatel GPS (192.168.43.180:3001)>: 2
[ WARN] [1582650333.614068009]: Novatel status code: 33554432
[ WARN] [1582650334.658189079]: Novatel status code: 33554432
[ WARN] [1582650335.658483239]: Novatel status code: 33554432
[ WARN] [1582650336.658574746]: Novatel status code: 33554432
```

pjreed, 2020-02-25:
This is the most immediately suspicious thing:
```
[ WARN] [1582650336.658574746]: Novatel status code: 33554432
```

Looking up the bits for that, it looks like that means your antenna is not powered and clock steering is disabled.  That might be ok if you've intentionally disabled those things -- although it may impact your receiver's ability to get its position -- but if not, you should try [enabling antenna power](https://docs.novatel.com/OEM7/Content/Commands/ANTENNAPOWER.htm) and [enabling clock steering](https://docs.novatel.com/OEM7/Content/Commands/CLOCKADJUST.htm).

As for these:
```
[ WARN] [1582650331.539180895]: publish rate failures detected <Novatel GPS (192.168.43.180:3001)>: 2
[ERROR] [1582650332.571151664]: insufficient data rate <Novatel GPS (192.168.43.180:3001)>: 4.000000 < 20.000000
```
If you only see this right after the driver starts, that error could be logged just because it hadn't yet collected enough messages to report accurate metrics.  It should probably be changed to just a warning message.  If you don't see those again and `rostopic hz` reports reasonable rates for the topics you're publishing, you can safely ignore those messages.

acelab-sungjin, 2021-05-21:
@pjreed Hello. I am working with novatel span cpt7, but I have some problems to use it. When I turned on the novatel_span_driver, the following warning occurs periodically. So the outpupt 'inspvax' provided jumped position data. How can I solve the problem?

`[ WARN] [1621596492.782138545]: publish rate failures detected <Novatel GPS (192.168.1.160:3001)>: 2
[ WARN] [1621596492.782866516]: Novatel status code: 50331648
[ WARN] [1621596493.782889759]: publish rate failures detected <Novatel GPS (192.168.1.160:3001)>: 2
[ WARN] [1621596493.783862158]: Novatel status code: 50331648
[ WARN] [1621596494.786698436]: publish rate failures detected <Novatel GPS (192.168.1.160:3001)>: 2
[ WARN] [1621596494.787494827]: Novatel status code: 50331648
[ WARN] [1621596495.791411477]: publish rate failures detected <Novatel GPS (192.168.1.160:3001)>: 2
[ WARN] [1621596495.792257502]: Novatel status code: 50331648
[ WARN] [1621596496.791552134]: publish rate failures detected <Novatel GPS (192.168.1.160:3001)>: 2
[ WARN] [1621596496.792338988]: Novatel status code: 50331648`

Thanks

acelab-sungjin, 2021-05-21:
I am using the following launch configuration.

`<?xml version="1.0"?>

<launch>
  <node name="novatel"
        pkg="nodelet" type="nodelet"
        args="standalone novatel_gps_driver/novatel_gps_nodelet">
    <rosparam>
      verbose: true
      connection_type: udp
      device: 192.168.1.160:3001
      polling_period: 0.05
      imu_sample_rate: 50
      use_binary_messages: true
      publish_novatel_positions: false
      publish_imu_messages: true
      publish_imu: true
      publish_novatel_velocity: false
      imu_frame_id: /imu
      publish_nmea_messages: false
      publish_novatel_heading2: false
      publish_novatel_dual_antenna_heading: false
      frame_id: /gps
    </rosparam>
  </node>
</launch>`

acelab-sungjin, 2021-05-21:
I'm using the following launch configuration

`<?xml version="1.0"?>

<launch>
  <node name="novatel"
        pkg="nodelet" type="nodelet"
        args="standalone novatel_gps_driver/novatel_gps_nodelet">
    <rosparam>
      verbose: true
      connection_type: udp
      device: 192.168.1.160:3001
      polling_period: 0.05
      imu_sample_rate: 50
      use_binary_messages: true
      publish_novatel_positions: false
      publish_imu_messages: true
      publish_imu: true
      publish_novatel_velocity: false<?xml version="1.0"?>

<launch>
  <node name="novatel"
        pkg="nodelet" type="nodelet"
        args="standalone novatel_gps_driver/novatel_gps_nodelet">
    <rosparam>
      verbose: true
      connection_type: udp
      device: 192.168.1.160:3001
      polling_period: 0.05
      imu_sample_rate: 50
      use_binary_messages: true
      publish_novatel_positions: false
      publish_imu_messages: true
      publish_imu: true
      publish_novatel_velocity: false
      imu_frame_id: /imu
      publish_nmea_messages: false
      publish_novatel_heading2: false
      publish_novatel_dual_antenna_heading: false
      frame_id: /gps
    </rosparam>
  </node>
</launch>
      imu_frame_id: /imu
      publish_nmea_messages: false
      publish_novatel_heading2: false
      publish_novatel_dual_antenna_heading: false
      frame_id: /gps
    </rosparam>
  </node>
</launch>`

acelab-sungjin, 2021-05-21:
I'm using the following launch configuration

<?xml version="1.0"?>

<launch>
  <node name="novatel"
        pkg="nodelet" type="nodelet"
        args="standalone novatel_gps_driver/novatel_gps_nodelet">
    <rosparam>
      verbose: true
      connection_type: udp
      device: 192.168.1.160:3001
      polling_period: 0.05
      imu_sample_rate: 50
      use_binary_messages: true
      publish_novatel_positions: false
      publish_imu_messages: true
      publish_imu: true
      publish_novatel_velocity: false
      imu_frame_id: /imu
      publish_nmea_messages: false
      publish_novatel_heading2: false
      publish_novatel_dual_antenna_heading: false
      frame_id: /gps
    </rosparam>
  </node>
</launch>

acelab-sungjin, 2021-05-21:
l am using the following launch configuration

```
<?xml version="1.0"?> 

<launch>
  <node name="novatel"
        pkg="nodelet" type="nodelet"
        args="standalone novatel_gps_driver/novatel_gps_nodelet">
    <rosparam>
      verbose: true
      connection_type: udp
      device: 192.168.1.160:3001
      polling_period: 0.05
      imu_sample_rate: 50
      use_binary_messages: true
      publish_novatel_positions: false
      publish_imu_messages: true
      publish_imu: true
      publish_novatel_velocity: false
      imu_frame_id: /imu
      publish_nmea_messages: false
      publish_novatel_heading2: false
      publish_novatel_dual_antenna_heading: false
      frame_id: /gps
    </rosparam>
  </node>
</launch>
```

danthony06, 2026-09-18:
Stale. There have been multiple fixes that address issues within this discussion, and it is not clear which specific issues remain.

closed 2026-09-18 by danthony06