swri-robotics/novatel_gps_driver #19: Does the driver take into account that the SPAN frame is different from the ROS frame?

JWhitleyAStuff, 2018-04-11:
The default SPAN frame of reference for Novatel is X to the right, Y forward, Z up. The default ROS frame is X forward, Y to the left, Z up. You can send VEHICLEBODYROTATION and APPLYVEHICLEBODYROTATION commands to the unit to correct for this (90 degrees about Z) but this isn't set by default. Does the driver take any of this into account?

pjreed, 2018-04-11:
The driver doesn't modify the values it gets from the IMU at all (aside from converting them into the right units for a sensor_msgs/Imu message), but it does have a param, `imu_frame_id`, that can be used to specify the `frame_id` of the Imu messages separately from the `frame_id` of other messages (which is usually irrelevant since fixes should always be in WGS84 coordinates).

The normal way of handling this in ROS is to set the `imu_frame_id` and then create a static transform publisher to connect that frame to the rest of your TF tree, then, inside your other nodes that consume Imu messages, transform them into the necessary frame.

pjreed, 2018-04-12:
I'll consider this to be closed by #20.

closed 2018-04-12 by pjreed