swri-robotics/novatel_gps_driver #25: Cannot find any data in the /insstdev topic.

sagniknitr, 2018-09-10:
The novatel driver is unable to log/parse data from the sensor into the /insstdev topic. Any help regarding this is highly appreciated.

pjreed, 2018-09-10:
What model device are you using, and what does your configuration for the node look like?  Is it correctly publishing other topics?

kelvin-sudani, 2018-09-10:
This is for ProPak6. It looks similar to #18.

pjreed, 2018-09-10:
I believe the INSSTDEV log actually does not exist on OEM6 devices; it was added in OEM7.  
For comparison, here's the OEM6 firmware reference manual: https://www.novatel.com/assets/Documents/Manuals/OM-20000144UM.pdf
And the OEM7 reference page is here:
https://docs.novatel.com/OEM7/Content/SPAN_Logs/OEM7_SPAN_Logs.htm?tocpath=Logs%7CView%20Logs%7CSPAN%20Logs%7C_____0

Depending on what data you need, you might be able to use the INSCOV log instead, or there might be another log that has it.

kelvin-sudani, 2018-09-10:
Essentially we are trying to get and `/odom` topic similar to the one from [novatel_span_driver: line 171-214](https://github.com/ros-drivers/novatel_span_driver/blob/master/novatel_span_driver/src/novatel_span_driver/publisher.py). Now for this we found all the neceasrry valuesd from other topics expect the `inspvax.pitch_std` , `inspvax.roll_std`, `inspvax.azimuth_std`.

pjreed, 2018-09-10:
INSCOV has an attitude covariance matrix, and I think you could extract the standard deviations you want from that.  The driver does something similar at https://github.com/swri-robotics/novatel_gps_driver/blob/master/novatel_gps_driver/src/novatel_gps.cpp#L935 to fill in the orientation covariance in IMU messages.

kelvin-sudani, 2018-09-10:
@pjreed Got it. Thanks.

pjreed, 2018-09-10:
Great, let me know if you need anything else.

closed 2018-09-10 by pjreed