swri-robotics/novatel_gps_driver #34: novatel_gps_msgs not built before novatel_gps_driver

msmcconnell, 2019-05-14:
When trying to build this package as part of a larger workspace using catkin_make, I got the following error repeatedly. 
```
In file included from /opt/carma/src/CARMANovatelGpsDriver/novatel_gps_driver/src/novatel_gps.cpp:35:0:
/opt/carma/src/CARMANovatelGpsDriver/novatel_gps_driver/include/novatel_gps_driver/novatel_gps.h:47:36: fatal error: novatel_gps_msgs/Gpgga.h: No such file or directory
compilation terminated.
CARMANovatelGpsDriver/novatel_gps_driver/CMakeFiles/novatel_gps_driver.dir/build.make:62: recipe for target 'CARMANovatelGpsDriver/novatel_gps_driver/CMakeFiles/novatel_gps_driver.dir/src/novatel_gps.cpp.o' failed
make[2]: *** [CARMANovatelGpsDriver/novatel_gps_driver/CMakeFiles/novatel_gps_driver.dir/src/novatel_gps.cpp.o] Error 1
CMakeFiles/Makefile2:22174: recipe for target 'CARMANovatelGpsDriver/novatel_gps_driver/CMakeFiles/novatel_gps_driver.dir/all' failed
make[1]: *** [CARMANovatelGpsDriver/novatel_gps_driver/CMakeFiles/novatel_gps_driver.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

This suggests to me that the novatel_gps_msgs were not built before the novatel_gps_driver. I found that if I built the novatel_gps_msgs package on its own first, then the novatel_gps_driver would build without issue. 

My development environment is Ubuntu 16.04, ROS Kinetic, cmake 3.5.1, gcc 5.4.0, Python 2.7.12 and Python 3.5.2

msmcconnell, 2019-05-14:
Adding 
```
add_dependencies(${PROJECT_NAME} ${catkin_EXPORTED_TARGETS})
```
To the CMakeLists.txt file in the novatel_gps_driver package resolved my issue. 

I can put up a PR unless the repo maintainers only wish to support catkin_tools instead of catkin_make

pjreed, 2019-05-14:
I'm surprised that that fails, actually; `novatel_gps_msgs` is passed into the `CATKIN_DEPENDS` list for `novatel_gps_driver`, which I thought would've been sufficient to make it build the msgs package first.  I only use catkin tools, but I've never seen any failures due to that in the official ROS build farm's builds.  Still, feel free to submit a PR if that fixes it for you.

MikeZhu92, 2019-06-03:
I come up with the same issue and @msmcconnell 's solution works for me too.

pjreed, 2019-06-04:
I noticed that you made a fix for it, so I'm going to go ahead and pull that in.  Thanks!

closed 2019-06-04 by pjreed

msmcconnell, 2019-06-04:
Glad to contribute!

PR #41 (2019-06-04): Add fix for messages build order
Fixes #34