swri-robotics/novatel_gps_driver #35: Error measurements not being filled out in fix message

danthony06, 2019-05-16:
Error measurements like the horizontal and vertical position accuracy are read from the receiver but not propagated to the GpsFix message that is published.

pjreed, 2019-11-19:
I finally took a look at this, and it turns out that the timestamps in GPGGA logs are actually in a different format from timestamps in the BESTPOS logs, but the driver is trying to parse them as though they're the same, and as a result it never manages to synchronize them.  I'll see about fixing that...

pjreed, 2019-11-19:
Actually, @danthony06 , it just occurred to me that I assumed you were talking about covariances, which is what I just fixed, but were you referring to other error values?

danthony06, 2019-11-20:
I was specifically looking at the err_horz and similar fields in the GPSFix message. In novatel_message_extractor.cpp around line 557 we note that we expect err_horz to be filled out from the BestPos message, but it doesn't look like we ever do so.

pjreed, 2019-11-20:
It doesn't seem like those exact values are reported in any of the Novatel messages, but I can calculate the DRMS and MRSE from the standard deviations provided in the BESTPOS log (see https://www.novatel.com/assets/Documents/Bulletins/apn029.pdf ) and fill those in there; is that what you'd expect?

danthony06, 2019-11-20:
Yes, that should do it.

PR #68 (2019-11-19): Fix covariance in GPSFix messages
The covariances in GPSFix messages were not being filled in
due to timestamps in GPGGA messages not being parsed properly.

Fixes #35
commit b8450165 (2019-11-19): Fix covariance in GPSFix messages (#68)

The covariances in GPSFix messages were not being filled in
due to timestamps in GPGGA messages not being parsed properly.

Fixes #35
PR #69 (2019-11-19): Fix covariance in GPSFix messages (Dashing)
The covariances in GPSFix messages were not being filled in
due to timestamps in GPGGA messages not being parsed properly.

Fixes #35
commit 34414034 (2019-11-19): Fix covariance in GPSFix messages (#69)

The covariances in GPSFix messages were not being filled in
due to timestamps in GPGGA messages not being parsed properly.

Fixes #35
closed 2019-11-20 by pjreed