swri-robotics/novatel_gps_driver #44: insufficient data rate <Novatel GPS (/dev/ttyUSB0)>: 0.000000 < 5.000000

yamfeng, 2019-06-19:
Hi ,I am using an OEM718D device that was running at 5 Hz and I have set the expected_rate_(5). When I use the Serial debugging assistant（cutecom），the data is successfully read, however when I use this novatel_gps_driver package, no data is obtained  through the commond ---roslaunch novatel_gps_driver tester_for_usb.launch ,and the error is :[ERROR] [1560927037.977802740]: insufficient data rate <Novatel GPS (/dev/ttyUSB0)>: 0.000000 < 5.000000
Is this package not support novatel 718D? or anything is wrong ? have you got the same problem with me?

pjreed, 2019-06-19:
I think that this error message is a known bug; see #6.  Although it prints that error message, it should still be able to publish GPS messages; do you see messages coming from your ROS topics?

yamfeng, 2019-06-20:
thankyou to your reply, I have already fixed this problem, it maybe that when I change the expected_rate_(5), the polling_period is forgoten to  be configured , and now I have set it to 0.5, the gps messages  is succesfully published, Is  this driver  compatible with OEM718D?

sharvil-patankar, 2019-11-05:
How did you change the expected rate? I couldn't find an API to change it. DId you change the source code? 
Thanks,

pjreed, 2019-11-05:
The value is currently hard-coded at https://github.com/swri-robotics/novatel_gps_driver/blob/master/novatel_gps_driver/src/nodelets/novatel_gps_nodelet.cpp#L206, but it might make sense to make this a separate parameter and/or have the default value be based on `polling_period`, which is a parameter.

danthony06, 2026-09-20:
PR #109 and https://github.com/swri-robotics/novatel_gps_driver/pull/153 addressed this.

closed 2026-09-20 by danthony06