swri-robotics/novatel_gps_driver #45: Driver unlogs all set logs (even the ones it doesn't set)

Staartvin, 2019-06-19:
To receive RTK corrections I set up my Novatel unit to send data to the NCOM1 port using **(LOG NCOM1 GPGGA ONTIME 5).** However, when I start the novatel_gps node it removes this log (and so my RTK corrections are not send anymore. 

This happens because in the Configure() method (see novatel_gps.cpp) the UNLOGALL command is performed. This clears all logs. Obviously, this is not what you want to do as it would remove all logs. I found out that I can set the _HOLD_ parameter for a LOG command to prevent it from being removed by UNLOGALL. However, this is more like a temporary fix than a permanent one, as using UNLOGALL is too evasive in my opinion.

Please fix?

pjreed, 2019-06-19:
This is intentional; the driver expects that it is the only thing accessing the port it is using to communicate with the device, and it disables all logs before reconfiguring them in order to ensure that the device is in a (semi-)reproducible state every time it runs.  In the past, we've seen bugs caused by logs that the driver tries to parse incorrectly, bandwidth issues due to too many things being logged at once, and the driver continuing to produce messages that users thought they had disabled; this helps to prevent those from happening.  (Note that there was recently a bug where logs were being removed on all ports, not just the one the driver is connected to; this was fixed in version `3.8.0`)

If you really do want a log to remain even when everything else on a port is reset, that is exactly what the `HOLD` parameter is intended for.  If it'd be useful for you to have the driver not unlog anything, though, I'd be fine with a PR that adds a config param that can optionally disable that.

Staartvin, 2019-06-19:
That makes sense. It seems I still run an old version that deletes all logs, not only the ones on the connected port. I'll upgrade to 3.8.0 and see if I feel like making a PR ;-) 

Thanks for your help!

danthony06, 2021-06-10:
I'm going to close this due to the intentional behavior and the long period of inactivity.

closed 2021-06-10 by danthony06