swri-robotics/novatel_gps_driver #49: Gpgga: Always sets was_last_gps_valid_ to true

punitagrawal, 2019-07-24:
https://github.com/swri-robotics/novatel_gps_driver/blob/d31d420ddfcc38ebf0139c99383856cba9360dc0/novatel_gps_driver/src/parsers/gpgga.cpp#L123

Gpgga parser seems to always set `was_last_gps_valid_` to true even in the case where the latitude and logitude data is reported to be empty.

pjreed, 2019-09-04:
Good catch, that does look like a bug.

pjreed, 2019-09-04:
Actually, after looking through this a bit more carefully, I think that the real bug here is the line above that which sets `was_last_gps_valid_` to false if the lat/lon are 0.  "valid" here means that the message was successfully parsed, which it was, even if the quality of the fix was invalid.  On the other hand, this seems to be related to #47, and it looks like the quality of the fix is not being used to indicate the status in the published message.

closed 2019-09-04 by pjreed

punitagrawal, 2019-09-04:
Thanks for taking a look at the report. 

Going by your description is a similar change needed in the GPRMC parser - https://github.com/swri-robotics/novatel_gps_driver/blob/bf2ba1756163dff28d3a0e81cf4cb45e5d53954b/novatel_gps_driver/src/parsers/gprmc.cpp#L127

It seems to me that GPRMC is using "was_last_gps_valid_" to indicate that the positions are valid rather than the success of message parsing.