swri-robotics/novatel_gps_driver #58: Inconsistent Get methods in the nodelet

sharvil-patankar, 2019-09-24:
Why are some get methods called without any if in front of it:
`gps_.GetGpggaMessages(gpgga_msgs);
 gps_.GetGprmcMessages(gprmc_msgs);
 gps_.GetNovatelPositions(position_msgs);
 gps_.GetNovatelXYZPositions(xyz_position_msgs);
 gps_.GetNovatelUtmPositions(utm_msgs);
 gps_.GetFixMessages(fix_msgs);
 gps_.GetNovatelHeading2Messages(heading2_msgs);
 gps_.GetNovatelDualAntennaHeadingMessages(dual_antenna_heading_msgs);
`

while others are called within an if statement:
`if (publish_imu_messages_)
      {
        std::vector<novatel_gps_msgs::NovatelCorrectedImuDataPtr> novatel_imu_msgs;
        gps_.GetNovatelCorrectedImuData(novatel_imu_msgs);
        for (const auto& msg : novatel_imu_msgs)
        {
          msg->header.stamp += sync_offset;
          msg->header.frame_id = imu_frame_id_;
          novatel_imu_pub_.publish(msg);
        }

        std::vector<sensor_msgs::ImuPtr> imu_msgs;
        gps_.GetImuMessages(imu_msgs);
        for (const auto& msg : imu_msgs)
        {
          msg->header.stamp += sync_offset;
          msg->header.frame_id = imu_frame_id_;
          imu_pub_.publish(msg);
        }

        std::vector<novatel_gps_msgs::InscovPtr> inscov_msgs;
        gps_.GetInscovMessages(inscov_msgs);
        for (const auto& msg : inscov_msgs)
        {
          msg->header.stamp += sync_offset;
          msg->header.frame_id = imu_frame_id_;
          inscov_pub_.publish(msg);
        }

        std::vector<novatel_gps_msgs::InspvaPtr> inspva_msgs;
        gps_.GetInspvaMessages(inspva_msgs);
        for (const auto& msg : inspva_msgs)
        {
          msg->header.stamp += sync_offset;
          msg->header.frame_id = imu_frame_id_;
          inspva_pub_.publish(msg);
        }
`
I am trying to add a new Get method but I don't know where to call it. I first thought that the defaults are loaded without the if condition but UTM is not the default and those messages are called without any if conditions. 
Thanks

pjreed, 2019-09-26:
Looking at that, I think that some of those slipped in by accident during merges.  A few of those are always pulled because they are used for generating diagnostics, but in general it's a good idea to only get messages if the node is going to actually publish them.  I'll clean up the others, but in the meantime you should put yours inside an `if` statement.

sharvil-patankar, 2019-09-26:
Thank you. I can close this issue now but let me know if you want it to stay open for your cleanup.

PR #59 (2019-10-01): Clean up message buffers
Some message buffers were being polled even if we weren't using them;
they've been cleaned up.  Also fixed a few compiler warnings.

Fixes #58
commit 2caa3fae (2019-10-01): Clean up message buffers (#59)

Some message buffers were being polled even if we weren't using them;
they've been cleaned up.  Also fixed a few compiler warnings.

Fixes #58
closed 2019-10-01 by pjreed