swri-robotics/novatel_gps_driver #62: INSSTDEV and INSCOV frequency

sharvil-patankar, 2019-11-05:
I noticed that the frequency for INSSTDEV and INSCOV was hardcoded to 1 in the source code. Is there a reason for this instead just making them the same frequency as the polling frequency?

pjreed, 2019-11-05:
INSSTDEV/INSCOV in particular are logged at 1 Hz because that is what the manual recommends for high-rate logging: https://www.scribd.com/document/413955753/Novatel-OEM7-Commands-Logs-Manual (see the documentation on the INSPVAX log)

For many logs, logging them at a higher rate would just consume more bandwidth but not actually produce significantly more precise data.  If there are particular logs you think would be useful at higher rates, it might make sense to make them configurable, but I'd check first to see whether they can actually change fast enough that it makes sense to log them at a higher rate.

Another thing to consider is that even with all of the logs that are currently at 1 Hz, just producing GPS & IMU using binary messages comes close to saturating an RS-232 connection, so you would need to be careful about how much more you enable.

sharvil-patankar, 2019-11-05:
I see, makes sense. I am posting a PDF from the novatel website for any future travellers who want the reference and don't want to get it from scribd. Go to the INSPVAX section.

https://docs.novatel.com/OEM7/Content/PDFs/OEM7_Commands_Logs_Manual.pdf

closed 2019-11-05 by sharvil-patankar