swri-robotics/novatel_gps_driver #78: DualAntennaHeading and Heading2 are not published (again)

RomanDmowski, 2020-01-23:
I have the latest version of novatel_gps_driver (Latest commit 3005b80 on Dec 5, 2019) compiled from source. I am running it on ROS Kinetic on Ubuntu 16.04. I have the Novatel PwrPak7D with Dual Antennas.
When I run the node, no information is being published to the DualAntennaHeading topic even though the DualAntennaHeading topic is listed. At the same moment, the log DUALANTENNAHEADINGA contains the right information with a heading value.

I use the lunch file:

<?xml version="1.0"?>
<launch>
  <node name="novatel"
        pkg="nodelet" type="nodelet"
        args="standalone novatel_gps_driver/novatel_gps_nodelet">
    <rosparam>
      verbose: false
      connection_type: tcp
      device: 192.168.100.221:2000
      use_binary_messages: false
      publish_novatel_dual_antenna_heading: true
      frame_id: /gps
    </rosparam>
  </node>
</launch>

I know there was similar issues in the past:

- Heading2 not being published #53
- Fix for publishing Heading2 and DualAntennaHeading information #57

It seems it still doesn't work.

ub-mri, 2020-01-31:
After a few tests, I found that If I added a log pointed at the driver port manually, the information (dual_antenna_heading) was published to the topic. So this is only partially a solution to that problem.

RomanDmowski, 2020-01-31:
After a few tests, I found that If I added a log pointed at the driver port manually, the information (dual_antenna_heading) was published to the topic. So this is only partially a solution to that problem.

pjreed, 2020-01-31:
Interesting -- when you say you pointed at the driver port manually, do you mean you used a serial connection instead of TCP?  It'd be pretty strange for one to work but not the other; does the issue still happen if you set `use_binary_messages` to `true`?  Also, would it be possible for you to record a pcap capture file of the network traffic and a log of the serial traffic so that I can compare them?

RomanDmowski, 2020-01-31:
I used Web UI to add log 'dualantennaheading' to the virtual port ICOM1 with the binary format. I changed use_binary_messages to 'true' in the launch file.

closed 2026-09-12 by danthony06