swri-robotics/novatel_gps_driver #80: dashing-devel error, GPGGA, GPRMC

Kyungpyo-Kim, 2020-02-28:
I really appreciate your effort!

I am trying to use ros2 and novatel pwrpak7, and I got some errors and warning.
To enhance this work, I'm reporting this error.

Later, I will use more and I can share the more result!

Thanks!

```bash
[component_container-2] [WARN] [novatel_gps]: For sentence: [GPGGA,,,,,,0,,,,,,,,]
[component_container-2] [ERROR] [novatel_gps]: Error reading from device <serial:/dev/ttyUSB1>: GPGGA log was invalid.
[component_container-2] [WARN] [novatel_gps]: Error parsing GPRMC message.
[component_container-2] [WARN] [novatel_gps]: For sentence: [GPRMC,,V,,,,,,,,,,N]
[component_container-2] [ERROR] [novatel_gps]: Error reading from device <serial:/dev/ttyUSB1>: Error parsing GPRMC message.
[component_container-2] [WARN] [novatel_gps]: GPGGA log was invalid.
[component_container-2] [WARN] [novatel_gps]: For sentence: [GPGGA,,,,,,0,,,,,,,,]
[component_container-2] [ERROR] [novatel_gps]: Error reading from device <serial:/dev/ttyUSB1>: GPGGA log was invalid.
[component_container-2] [WARN] [novatel_gps]: Error parsing GPRMC message.
[component_container-2] [WARN] [novatel_gps]: For sentence: [GPRMC,,V,,,,,,,,,,N]
[component_container-2] [ERROR] [novatel_gps]: Error reading from device <serial:/dev/ttyUSB1>: Error parsing GPRMC message.
[component_container-2] [WARN] [novatel_gps]: GPGGA log was invalid.
[component_container-2] [WARN] [novatel_gps]: For sentence: [GPGGA,,,,,,0,,,,,,,,]
[component_container-2] [ERROR] [novatel_gps]: Error reading from device <serial:/dev/ttyUSB1>: 
GPGGA log was invalid.
[component_container-2] [WARN] [novatel_gps]: Error parsing GPRMC message.
[component_container-2] [WARN] [novatel_gps]: For sentence: [GPRMC,,V,,,,,,,,,,N]
[component_container-2] [ERROR] [novatel_gps]: Error reading from device <serial:/dev/ttyUSB1>: Error parsing GPRMC message.
[component_container-2] [WARN] [novatel_gps]: GPGGA log was invalid.
[component_container-2] [WARN] [novatel_gps]: For sentence: [GPGGA,,,,,,0,,,,,,,,]
[component_container-2] [ERROR] [novatel_gps]: Error reading from device <serial:/dev/ttyUSB1>: GPGGA log was invalid.
```

pjreed, 2020-02-28:
Thanks for the report.  Usually when a device is getting blank messages like `GPGGA,,,,,,0,,,,,,,,`, it's because it's not able to track any satellites; do you have an antenna connected, and does the antenna have a clear view of the sky?

You can also echo the output of the `diagnostics` topic to see if it has detected any hardware errors, and the output from the `bestpos` topic should show how many satellites it is tracking.

Kyungpyo-Kim, 2020-03-13:
> Thanks for the report. Usually when a device is getting blank messages like `GPGGA,,,,,,0,,,,,,,,`, it's because it's not able to track any satellites; do you have an antenna connected, and does the [...]

I checked the antenna and it worked at outside. However, there was no bestpos topic. To get the bestpos topic, do I have to set something of the default one?

This is my setting for your reference.
```yaml
    # Novatel # 
    novatel_container = launch_ros.actions.ComposableNodeContainer(
        node_name='novatel_gps_container',
        node_namespace='',
        package='rclcpp_components',
        node_executable='component_container',
        composable_node_descriptions=[
            launch_ros.descriptions.ComposableNode(
                package='novatel_gps_driver',
                node_plugin='novatel_gps_driver::NovatelGpsNode',
                node_name='novatel_gps',
                parameters=[{
                    'connection_type': 'serial',
                    'device': '/dev/ttyUSB1',
                    'verbose': True,
                    'publish_novatel_positions': True,
                    'frame_id': '/gps'
                }]
            )
        ],
        output='screen'
    )
```

Kyungpyo-Kim, 2020-03-13:
I found bestpos topic. It's my mistake with setup environment. Thanks!

closed 2020-03-13 by Kyungpyo-Kim