swri-robotics/novatel_gps_driver #82: sensor_msgs/Imu pre-transformed

msmcconnell, 2020-03-06:
I have noticed that the orientation in the sensor_msgs/Imu output is being negated and the roll/pitch value usage not being used to match the INSPVA log definition. 
```
      imu->orientation = tf::createQuaternionMsgFromRollPitchYaw(inspva->roll * DEGREES_TO_RADIANS,
                                              -(inspva->pitch) * DEGREES_TO_RADIANS,
                                              -(inspva->azimuth) * DEGREES_TO_RADIANS);
```
In the above code snippet from novatel_gps.cpp you can see that the pitch and azimuth values are negated. Additionally, in the tf library roll corresponds to rotation around the x axis while pitch corresponds to rotation around the y axis. However in the [INSPVA doc](https://docs.novatel.com/OEM7/Content/SPAN_Logs/INSPVA.htm) you can see that the roll is actually coming from the vehicle frame Y-axis and the pitch is coming from the vehicle frame X-axis. 

It seems like this code might be trying to make the imu match the ROS convention of X being forward and Y being to the right, but I'm not sure how the -Z value is coming into play. Could someone elaborate on the motivation here?

pjreed, 2020-03-06:
I believe @JWhitleyWork originally wrote that code, so he might be able to elaborate on it.

JWhitleyWork, 2020-03-09:
Unfortunately, I did not write the code, I'm just a lowly maintainer. I'll look into this a bit further, though, and see if I can figure out what's going on.

danthony06, 2026-09-12:
Closing as stale.

closed 2026-09-12 by danthony06