swri-robotics/novatel_gps_driver #84: boost::thread sleep not working

zang09, 2020-04-06:
I update my ros driver, after that this driver not publishing any ros topics!
Then, I figure out that while loop in Spin() function execute only first time.
So I change the code:

(+) ros::Rate loop_rate(1000);

while(gps_.IsConnected() && ros::ok()) {
  (-) boost::this_thread::sleep(boost::posix_time::microseconds(1));
  (+) loop_rate.sleep();
}

Then it finally publish the topics again!
Please fix this errors.

pjreed, 2020-04-08:
When you say it only executes the first time, what happens after that?  Is something throwing an exception, or is it blocking somewhere?  Has anybody else seen this happen?

It seems especially odd that using ros::Rate there would make a difference; one of the differences in using a ROS timer is that it follows ROS's clock when running in a simulation, and, if anything, I would expect that to cause problems since the timer will stop running if the simulation is paused.  Normally that's not an issue for hardware drivers since you don't usually hardware drivers in simulation, though.

zang09, 2020-04-09:
In my case after enter the command like this, 
$ sudo apt-get update && sudo apt-get upgrade
I can see same problem in all my laptop and mini PC(NUC)

When I debugging the code, there was no execption but the code is block at below position.
-> boost::this_thread::sleep(boost::posix_time::microseconds(1));

Then, I launch the driver in real situation not with simulation.
My computer enviornment is: Ubuntu 18.04 && ROS Melodic

cbradyas, 2020-04-22:
We also had this problem. I made that  change and it fixed it.

Kyungpyo-Kim, 2020-04-24:
I also got same problem.
My computer environment is: Ubuntu 18.04 && ROS Melodic

closed 2020-04-27 by pjreed

pjreed, 2020-04-27:
How strange, I had never seen this problem before, but I tested it on a different computer and now I see it there.  Well, I merged in a change that should fix it everywhere.  Thanks!

PR #85 (2020-04-27): Use ros::WallRate to sleep instead of boost
This fixes #84, where some users were seeing boost sleep forever.