swri-robotics/novatel_gps_driver #87: /fix and /imu have different time stamps

HHF111, 2020-05-31:
The timestamps of /fix and /imu ros topics are far away from each other. How to solve that? Thanks!

pjreed, 2020-06-03:
The first thing you should do is check your computer's clock and make sure it's synchronized with a GPS time source somewhere.  The timestamps in the headers for sensor_msgs::NavSatFix and sensor_msgs::Imu messages are based on the computer's clock when the message is generated.  They can be adjusted if something is publishing std_msgs::Time messages to `/gps_sync`, but that's unlikely if you're not aware of it.

The timestamps in `gps_common/GPSFix` messages under `status.header.stamp` are based on the actual GPS clock, though, so they could differ if your computer's clock is not synced.

If your computer clock is synced and you're seeing the timestamps in the message headers be far away from each other, that could be a bug; could you record a serial capture log or a pcap file and a bag file with some example data and upload them?

danthony06, 2026-09-18:
stale.

closed 2026-09-18 by danthony06