swri-robotics/novatel_gps_driver #88: How do I set the SETINSTRANSLATION and SETINSROTATION in Linux?

manasmacherla, 2020-05-31:
Hi,

While working in Linux how do I set the translational offsets of the antenna to the receiver (Powerpak 7d-e1)??

pjreed, 2020-06-03:
Right now this driver doesn't have a way to directly set those values.  The best way to set those right now would be to use something like NovAtel Connect in Windows or a serial terminal emulator like cutecom in Linux to connect to a serial port, manually run the `SETINSTRANSLATION` and `SETINSROTATION` commands, and then start the driver.

I think a reasonable change to the driver would be to add a section to its config params to define an IMU configuration and then have the driver send those commands on startup; alternately, a more generic way to handle that kind of thing would be to add a service that lets a user run an arbitrary command and then display the result.

manasmacherla, 2020-06-03:
If possible, can this be implemented in the driver??

pjreed, 2020-06-03:
This seems useful, so I'd like to implement it, but I'm pretty busy right now and I don't have access to an INS device, so it might be a while before I can get to it.

If anybody else wants to take a look at it, I think this would be pretty straightforward; I would add new parameters to set the INS rotation/translation values and then set those in the NovatelGps::Configure function on startup.

manasmacherla, 2020-06-04:
I will try to do this and get back to you if I face any issues. Thank you..

closed 2026-09-12 by danthony06