swri-robotics/novatel_gps_driver #91: Nothing published on rostopics

Smiffe, 2020-06-16:
Hi!
I'm using Flexpak6 device and I noticed that's not supporting "RAWIMUX" command. Anyway ROS starting without any errors, warnings or infos. There are available rostopics, but nothing is published. I checked serial communication and all frames are going trough.

Is that missing command responsible for this situation?

pjreed, 2020-06-29:
Just to be sure, do you have a SPAN device?  Not all FlexPak6 enclosures have SPAN support, and if you don't, then it won't produce any IMU-related messages.

If you connect directly to it with a serial terminal (I like [CuteCom](http://cutecom.sourceforge.net/) ), if you type something like `log rawimuxa once`, does it print anything?

Smiffe, 2020-07-07:
Yeah, it's not SPAN device so it doesn't support "RAWIMUX" command (it prints back info about that after sending a command).

The main problem was empty rostopics, but after some tries i fixed it. The first problem was GPS rate lower than 20Hz described in for example #6 or #44 issue and the second was with antenna. Testing device was in garage and it wasn't getting fix. I was getting "insufficient data rate" error even with all frames visible on serial port so I thought it's ros/driver problem, not device problem.

It would be easier if driver could parse messages with no fix inside and I could see something when doing "rostopic echo" for example "no fix" when tracking 0 satellites. Anyway it's not any bug, driver works fine, so I'm closing this Issue.

closed 2020-07-07 by Smiffe