swri-robotics/novatel_gps_driver #93: A question about NovAtel INSPVA message to RPY

DarrenWong, 2020-06-29:
Trying about parser  NovAtel INSPVA message to roll pitch yaw.
 According to  https://docs.novatel.com/OEM7/Content/SPAN_Logs/INSPVA.htm, the azimuth is left-hand rotation so that it add should negative symbol. My question is both roll and pitch are right-hand rotation, why pitch needs to add negative as  **-(inspva->pitch) * DEGREES_TO_RADIANS**?  

 Thanks a lot!
[image]

https://github.com/swri-robotics/novatel_gps_driver/blob/98d27ba95651b8c89a37fb5a431cfab1b7aa9d1c/novatel_gps_driver/src/novatel_gps.cpp#L968
```
     imu->orientation = tf::createQuaternionMsgFromRollPitchYaw(inspva->roll * DEGREES_TO_RADIANS,
                                              -(inspva->pitch) * DEGREES_TO_RADIANS,
                                              -(inspva->azimuth) * DEGREES_TO_RADIANS);
```

pjreed, 2020-06-29:
It appears like the code there was changed to that from this:
```cpp
      imu->orientation = tf::createQuaternionMsgFromRollPitchYaw(inspva->roll * DEGREES_TO_RADIANS,
                                              inspva->pitch * DEGREES_TO_RADIANS,
                                              (90.0 - inspva->azimuth) * DEGREES_TO_RADIANS);
```

By @JWhitleyWork in 603ff6be in order to match results obtained via experimental testing.  Does that produce incorrect output for you?  Unfortunately, I don't have a SPAN device I can actually test this with.

DarrenWong, 2020-06-30:
Thanks @pjreed  
I have a bag file that contains inspva message and the SPAN device is aligned with the system well to set as ground truth.
Nott sure the correct way for roll and pitch convert to euler angle. Might be I get the device to test the result is better.

Also read some code from apolloauto is similar to the previous one that adds negative to pitch 
https://github.com/ApolloAuto/apollo/blob/01ce8b2831f466d3dd48ed7b119aff8a2156436d/modules/drivers/gnss/parser/novatel_parser.cc#L681

```
// Converts NovAtel's azimuth (north = 0, east = 90) to FLU yaw (east = 0, north
// = pi/2).
constexpr double azimuth_deg_to_yaw_rad(double azimuth) {
  return (90.0 - azimuth) * DEG_TO_RAD;
}

  ins_.mutable_euler_angles()->set_x(pva->roll * DEG_TO_RAD);
  ins_.mutable_euler_angles()->set_y(-pva->pitch * DEG_TO_RAD);
  ins_.mutable_euler_angles()->set_z(azimuth_deg_to_yaw_rad(pva->azimuth));
```

closed 2020-06-30 by DarrenWong