swri-robotics/novatel_gps_driver #97: Novatel RTCM3 Correction Support

cagataysari, 2020-10-08:
Hello, 

Is RTCM3 supported by the driver ? 

I have a related question related to Novatel IGM SPAN A1
I am trying to get the driver working on a nvidia's machine. The problem is, when I connect novatel IG to the machine via USB (USB Cable ) , virtual ports are not detected. So I try to configure User Port by connecting it to the machine via serial adapter. 
The configuration contains the followings

       log thisport gpgga ontime 0.05
       log thisport gprmc ontime 0.05
       log thisport bestposa ontime 0.05
       interfacemode thisport RTCMV3 novatel

then I run tester_for_usb.launch file, I keep getting warning regarding to frequecny, Expected one is: 20 hz, I get around 2~3 Hz and also correction code messes up the parser, How can I fix this ?

Best Regards

pjreed, 2020-10-09:
This may or may not be related to your problem, but something I've seen before is that some nvidia device firmwares do not come with the cdc-acm module installed by default; the driver needs to be able to connect to the device as an ACM module (e. g., `/dev/ttyACM0`), and if that's not present, this might be useful for you: https://github.com/jetsonhacks/installACMModule

I'm not familiar with RTCMV3, but it looks like that message format is a little different than the Novatel message format, and so I suspect it won't work; the message parsers for the relevant message types will probably need to be modified so that they can understand RTCMV3 data.

cagataysari, 2020-10-12:
Hello pjreed,

cdc-acm and other drivers are there. I also use FTDI based usb over the serial connect to User Port. But I can see FTDI serial is there not problems. The issue is with USB port, USB port's virtual serial port are not shown when I dmesg. I also ask this on Nvidia's Drive AGX forum. By the way Drivework's Novatel Driver works fine on USB but I wanna use your driver package since kernel won't detect them I can't use it. I try to configure another port, I have a frequency issue. I will try some other ways to get your driver working on Drive AGX. I will write them here maybe somebody else might use it too.

danthony06, 2026-09-20:
Fixed via https://github.com/swri-robotics/novatel_gps_driver/pull/160.

closed 2026-09-20 by danthony06