swri-robotics/novatel_gps_driver #99: How does the `novatel_gps_node.cpp` source file generate?

zzzzzzli, 2020-11-17:
There is no main function in any source file in the directory `novatel_gps_driver/src/` , but when I build it with `catkin_make`, a main function source file named `novatel_gps_node.cpp` generates in the `build/novatel_gps_driver/
` directory, so does the corresponding executable `novatel_gps_node`. I wonder where does the file  `novatel_gps_node.cpp` come from. Thank you!

danthony06, 2020-11-17:
I believe in https://github.com/swri-robotics/novatel_gps_driver/blob/master/novatel_gps_driver/CMakeLists.txt, the ```swri_nodelet_add_node()``` call on line 97 generates this during compilation.

pjreed, 2020-11-17:
That's right; the driver is intended to be primarily used as a nodelet, and that's defined in `novatel_gps_driver/src/nodelets/novatel_gps_nodelet.cpp`.  The `swri_nodelet_add_node` CMake macro is a convenience function that generates a standalone node executable from a template (https://github.com/swri-robotics/marti_common/blob/master/swri_nodelet/nodelet.cpp.in), which is primarily useful for testing and debugging purposes.

zzzzzzli, 2020-11-18:
OK, thank you very much.

closed 2020-11-18 by pjreed