swri-robotics/novatel_gps_driver #100: Is this package available for GNSS not only GPS?

nicolasaw, 2020-12-03:
Hi guys, I am working on a project and my GNSS receiver module somehow was only able to show Glonass NMEA Message. This makes me unable to use the package because it requested for NMEA Message from GPS. Do I need to change something in the launch file or is there something I can do with my GNSS receiver? I am using U-Blox F9P. Thanks in advance

pjreed, 2020-12-03:
In theory this driver should work with a NovAtel receiver that doesn't have a GPS, but I probably wouldn't expect it to work with any non-NovAtel device.  This driver looks like it's supposed to work with U-Blox devices, so you might give it a try: https://github.com/KumarRobotics/ublox

kriskozak, 2020-12-03:
@nicolasaw, is this the repo that you intended for this issue? This repo is for a ROS driver specifically made to work with (some) Novatel GNSS receivers. If so, and because you mention that you are using a U-Blox receiver, I'd recommend that you have a look at one of the U-Blox specific drivers. For example:
* https://github.com/KumarRobotics/ublox
* https://github.com/ros-agriculture/ublox_f9p

kriskozak, 2020-12-03:
Jinx, @pjreed

pjreed, 2020-12-03:
Darn!

nicolasaw, 2020-12-04:
Hi thanks for the feedback!! I will surely look up on those packages. I also opened an issue in here https://github.com/swri-robotics/mapviz/issues/712. It is related to ROS Foxy as I was not able to fully used Mapviz. Is there any comments about it as well? Thanks in advance!

closed 2020-12-04 by pjreed