swri-robotics/novatel_gps_driver #101: GPSFix track source improvements

ozzdemir, 2021-01-15:
I was comparing track angle from bestvel and inspvax and got the following result. Is there a specific reason driver uses track angle from bestvel instead of inspvax? I was curious about it since track from bestvel is not stable at low speeds.
[image]

pjreed, 2021-01-15:
Hmm.  I'm surprised that bestvel is not stable at low speeds, because it's supposed to be the best velocity estimate from all combined sources (https://docs.novatel.com/oem7/Content/Logs/BESTVEL.htm).  The reason not to just use inspvax is because that log isn't available at all on devices that don't have an IMU.  There might be an option somewhere that needs to be set to factor that into bestval calculation... if not, it'd probably be best to use inspvax.azimuth if available and fall back to bestvel.track_ground if it's not.

ozzdemir, 2021-01-15:
In the end of the documentation you shared, the following statement exists;
```
Velocity (speed and direction) calculations are computed from either Doppler or carrier phase measurements rather than from pseudorange measurements. Typical speed accuracies are around 0.03m/s (0.07 mph, 0.06 knots).

Direction accuracy is derived as a function of the vehicle speed. A simple approach would be to assume a worst case 0.03 m/s cross-track velocity that would yield a direction error function something like:

d (speed) = tan-1(0.03/speed)
```
So, it seems to be not using all the sources, it calculates direction from velocity measurements.

pjreed, 2021-01-15:
Oops, that's what I get for just skimming over the description.

Unfortunately, I haven't yet found a single "best" source of orientation, and in fact there's three -- [inspvax](https://docs.novatel.com/oem7/Content/SPAN_Logs/INSPVAX.htm) for SPAN receivers, [heading2](https://docs.novatel.com/oem7/Content/Logs/HEADING2.htm) for ALIGN receivers, and [bestvel](https://docs.novatel.com/oem7/Content/Logs/BESTVEL.htm) for everybody else.  So the code should probably figure out which one of those are available and use the best one...

danthony06, 2026-09-12:
Fixed via https://github.com/swri-robotics/novatel_gps_driver/pull/141.

closed 2026-09-12 by danthony06