swri-robotics/novatel_gps_driver #104: Launching the driver for multiple sensors

uchaudh, 2021-03-13:
I need some help to launch the ros2 driver for 2 receivers simultaneously. I changed the node_namespace in both the launch files but I don’t think the driver is built in a way that it takes this namespace and publishes the topics under it. It creates two separate containers but the drivers end up publishing to the same topic names. I need to publish them under different namespaces, say for example /driver1/bestpos and /driver2/bestpos.

m4rk0401, 2021-03-13:
Hi @uchaudh, I can't help you directly with your problem, but have a question about launching the .launch file. Can you maybe answer a question for me?

uchaudh, 2021-03-13:
Sure, what is the question?

Ms. Utkarsha Mukund Chaudhari
Graduate Student - MS Automotive Engineering
Clemson University - ICAR
Ph No : +1 (864) 887-9693

pjreed, 2021-04-05:
Sorry for the delayed response.  I think that it should work the way you're expecting; could you upload your launch files so we can take a look at them?

One thing to note is that I suspect it may not work properly if you are trying to run two instances of the same nodelet in a single nodelet container; you probably have to run two separate nodes for it to work that way.

uchaudh, 2021-04-05:
Thank you for your response. I was able to solve the issue.

closed 2021-06-10 by danthony06