swri-robotics/novatel_gps_driver #107: Very Low publish rate for Novatel Propak 6 with RTK enabled on ROS Melodic

hangcui1201, 2022-01-20:
Hello, 

Our team is using Novatel Propak 6 with RTK correction service from Hexagon SmartNet. After enable the RTK correction service, the computed solution became "NARROW_INT". However, the publish rate also became pretty low, around 2. 

The message I got is 
[WARN] [xxxx.xxxx]: publish rate failures detected <Novatel GPS (192.168.74.10:2000)>: 2>

Is there any way to improve this?

Thank you very much!

Best,
Hang

danthony06, 2022-03-02:
I don't have a Propak 6 to test with. I'm not sure if someone with more experience with these devices might have some ideas on how to troubleshoot this.

danthony06, 2026-09-08:
Closing this since Melodic is long since EOL.

closed 2026-09-08 by danthony06