swri-robotics/novatel_gps_driver #112: ROS2 Galactic support?

josephlu-robot, 2022-07-21:
Hi, 
I saw the branch dashing-level which support dashing in ROS2. But seems that I can't build it from source for my ROS2 galactic environment, can anyone help with this? Or is it actually not supporting yet? Thanks.

danthony06, 2022-07-22:
Sorry, I missed releasing this for Galactic. We definitely plan on releasing it.

danthony06, 2022-08-01:
By the way, you should be able to build the dashing-devel branch from source in Galactic. I just did it on a clean installation, and it seemed to work fine.

danthony06, 2022-08-03:
It's available in Foxy and Galactic. If you have a specific build error, please open an issue with the specific problem, because it appears to be working.

https://github.com/ros/rosdistro/blob/0cb60362e83c1431348f68baa5e893107559f961/galactic/distribution.yaml#L2559

closed 2022-08-03 by danthony06