swri-robotics/novatel_gps_driver #119: New release into Rolling

clalancette, 2024-03-18:
Hello!

It looks like novatel_gps_driver is failing to build in Rolling: https://build.ros2.org/view/Rbin_uN64/job/Rbin_uN64__novatel_gps_driver__ubuntu_noble_amd64__binary/ .

The good news is that it looks like this is already fixed in the sources by #116 .  Could you please do a new release of this package into at least Rolling, which will fix the error?  Thank you!

danthony06, 2024-03-18:
@clalancette Thanks, I released 4.1.2 for Humble, Iron, and Rolling.

https://github.com/ros/rosdistro/pull/40246
https://github.com/ros/rosdistro/pull/40247
https://github.com/ros/rosdistro/pull/40248

closed 2024-03-18 by danthony06

clalancette, 2024-03-18:
> @clalancette Thanks, I released 4.1.2 for Humble, Iron, and Rolling.

Thank you!